forsyde/forsyde-deep #32: GHC 8.0 Compatibility

HWoidt, 2016-05-24:
GHC 8.0.1 is out. A cursory glance at the migration guide did not raise any alarms with me but we still need to check, whether everything works. There are also some new features that might be interesting (although we should not introduce any changes that break backwards compatibility for GHC 7.10):

 * A more refined interface for implicit call-stacks, allowing libraries to
   provide more helpful runtime error messages to users

 * Significant improvements in error message readability and content, including
   facilities for libraries to provide custom error messages, more aggressive
   warnings for fragile rewrite rules, and more helpful errors for missing
   imports

 * Introduction of type application syntax, reducing the need for proxies

https://downloads.haskell.org/~ghc/8.0.1/docs/html/users_guide/8.0.1-notes.html
https://ghc.haskell.org/trac/ghc/wiki/Migration/8.0

ingo-sander, 2016-05-24:
We could already now run the deep-embedded code through Travis CI. You can simply copy (and possible extend) my travis.yml file, which I have used in [forsyde-shallow] (https://github.com/forsyde/forsyde-shallow). It builds the packages for both 7.10.3 and 8.0.1 and sends a report, if something is wrong. So far there are no tests included.

ugeorge, 2016-06-14:
While trying to test forsyde-atom through TravisCI I've noticed that type-level already fails with ghc-8.0.1. Since ForSyDe-Deep is also dependent on type-level, I imagine the migration won't be so straightforward. I haven't checked it in detail though.

HWoidt, 2016-07-10:
Fixed through https://github.com/forsyde/forsyde-deep/pull/33

closed 2016-07-10 by HWoidt